M1z23R/ESP8266-EvilTwin #3: next update

MRFOX777, 2020-08-18:
When will the next update come?
Please improve ui and fix bug.

MRFOX777, 2020-11-19:
Hello brother.  When will the next update come?

TuserAhammed, 2022-01-25:
How to edit bin file ,make a tutorial

cakralithium, 2022-02-06:
> Cara mengedit file bin, buat tutorial

file bin tidak mungkin bisa diedit kecuali file ino.. edit dan compile ulang

cakralithium, 2022-02-06:
> Cara memasukkan file bin, buat tutorial

flash pakai flasher 32bit atau 64 bit tergantung pc kamu... sebelum di flash. instal dulu esp8266 agar terbaca di pc..mu

cakralithium, 2022-02-07:
> Saya baru saja merilis v2, silakan coba dengan v2. Sayangnya UI lebih buruk - ini sangat mendasar. Jangan ragu untuk mengedit string yang berisi kode html secara manual.

> Saya baru saja merilis v2, silakan coba dengan v2. Sayangnya UI lebih buruk - ini sangat mendasar. Jangan ragu untuk mengedit string yang berisi kode html secara manual.

apakah ada pengembangan untulk penambahan button dan oled.. program ini sudah bekerja dengan baik.. mungkin dengan penambahan itu akam lebih baik lagi.. thanks