M1z23R/ESP8266-EvilTwin #4: I need some help!

alari73512, 2020-09-15:
Hi! Im a complete noob in this stuff, but i think that it would be awesome if you made a step by step guide for the beginners

Many thanks ! :)

alari73512, 2020-09-15:
Or if you could help me through teamviewer :)

M1z23R, 2020-09-20:
What help do you need?

ShahriarShafin, 2020-09-22:
@M1z23R I can't compile. Do you have a bin file?

M1z23R, 2020-09-23:
@ShahriarShafin 
https://github.com/M1z23R/ESP8266-EvilTwin/releases

alari73512, 2020-09-24:
Basically i need help with everything like a step by step guide with pictures or like could you record yourself doing the process? i would just like to see how you do it from the start till the end. Thank you!

M1z23R, 2020-09-26:
I've written an entire step by step tutorial on the main project page: https://github.com/M1z23R/ESP8266-EvilTwin
Just scroll down

alari73512, 2020-09-27:
M1z23R. yeah i figured it out, firsti time using arduino. Tell me this, Will you update this project? or like atleast make the captive portal of the evil twin a little more realistic? thank you in advance. Its an awesome project tbh.

M1z23R, 2020-09-27:
@alari73512 I will, at the moment I have a lot of collage duties but I'll try to make time to redo the project with better UI and some bug fixing.

gvoze32, 2020-10-04:
@M1z23R the evil twin is working perfectly but deauther isn't working, can you fix it?

adolabsnet, 2020-10-04:
P

ewinz19, 2021-04-10:
this is just a trick...hahahaha....I'm sorry ... because my city is very dearly valued