M1z23R/ESP8266-EvilTwin #8: can we have a vid of this ? also i have a lot of ideas/questions

minanagehsalalma, 2020-11-24:
first wouldn't this need two ESPs ? one to deauth and one to host the ap  and web server? 

second how does it handle the attack or we do need an esp32 ?

third how do you verify the pass ? i have a hella better way to do it offline ( without even needing aircrack or such)

@M1z23R will be waiting for your reply ::)

M1z23R, 2020-11-24:
Thank you for the feedback, well in short, ESP supports duo mode (wifi ap+client) at the same time. AP hosts the fake AP and client is deauthing until a password arrives. For a moment it stops deauthing and tries to connect to the original AP using the given password. If connection is successful it knows that the password is correct and stop everything and returns to original "admin" hidden AP. I hope I was clear enough. Let me know about the offline cracking, I look forward to it.

minanagehsalalma, 2020-11-24:
@M1z23R how many devices can it handle in the ap ? 
like what if 5 devices connect to it will it crash ? 

the offline cracking is using js in the browser so it's up to the client device not the esp (means it won't make any additional loads on our esp)

https://github.com/derv82/capjs

M1z23R, 2020-11-24:
Haven't really checked the number of devices, but I doubt more then 3 people will be connected at the same time (one being admin, and one or two at most "victims"). And regarding the capjs, it's not possible since the project doesn't capture packets, first due to lack of storage for them, and 2nd due to it being hard to implement.

minanagehsalalma, 2020-11-25:
> it's not possible since the project doesn't capture packets

@M1z23R  i know but it can ... and also handshakes are super small.

> and 2nd due to it being hard to implement.

The capjs or the packet capture? 

If so , space huhn have already made one that's a bit indirect.

https://github.com/spacehuhn/ArduinoPcap

minanagehsalalma, 2020-11-25:
> first due to lack of storage for them,

@M1z23R will ESP32 be a fix to this problem ?

btw handshake files are in the size of bytes mostly.

M1z23R, 2021-10-20:
It's pretty much the same - at least the concept.
Changes are: Deauther now actually works cuz i forgot to add channel switching for deauth in v1.
Web UI is more basic/uglier in v2
Web Server (Async) used in v1 was a bit unstable so I switched back to normal espresiff web Server.
SPIFFS was removed due to it being buggy so no more saved logs after reseting the board.