M1z23R/ESP8266-EvilTwin #16: Deauther Not Working

keko01, 2021-05-28:
First of all, health to your labor. Unfortunately, the deauther feature, which is the most important feature, does not work. Good work mate, looking forward to this one to be resolved first and a new one.

M1z23R, 2021-10-19:
I've just released v2, feel free to try with v2.
In v1, I forgot to add a switch for channel so probably only APs on channel 1 could get deauthed.

AsherPhan, 2021-11-23:
Although channel hopping is working and I verified, the Deauther still not working

adierc, 2021-12-07:
if use original board driver, deauth wont work, i use driver from another source it work just fine

cakralithium, 2021-12-09:
deauther and evil twin its work..
enter wifi server M1z23R 
push start deauthing and start evil twin
sign in wifi fake.. and  192.168.4.1..and wait for  got password ... stop deauthing