M1z23R/ESP8266-EvilTwin #30: Evil Twin Not turn off after giving right password

Kuntalchanda, 2022-01-23:
Hey  I'm checking this project on my own router, 
deauthentication attack working great, but issues with EvilTwin,  after giving right Wi-Fi password that showing "WRONG PASSWORD" but AP was connected to my ROUTER ( checking my router DHCP client)   , but at your Given instranctions ESP8266 not  restarted with default ssid M1z23R / deauther and so can't check the stored password !!! , please fix this issue

cakralithium, 2022-02-08:
add intruksi deauthing false after get right password

giadat, 2022-10-21:
> thêm intruksi deauthing false sau khi có đúng mật khẩu

What is the order of the line in the code table bro?

MrDrak045, 2024-12-03:
Try this code 
from scapy.all import *
import os

# Define the legitimate network details
trusted_ssid = "YourNetworkName"
trusted_bssid = "XX:XX:XX:XX:XX:XX"  # Replace with your AP's BSSID (MAC address)

def detect_evil_twin(packet):
    # Look for beacon frames (type 0, subtype 8)
    if packet.haslayer(Dot11Beacon):
        ssid = packet[Dot11Elt].info.decode()  # Get SSID
        bssid = packet[Dot11].addr2  # Get BSSID (MAC address)

        if ssid == trusted_ssid and bssid != trusted_bssid:
            print(f"[ALERT] Evil Twin Detected! SSID: {ssid}, Fake BSSID: {bssid}")

            # Optional: Disable Wi-Fi to stop further connection
            os.system("nmcli radio wifi off")
            print("Wi-Fi disabled for protection.")

# Sniff packets on the default Wi-Fi interface
try:
    print("Scanning for Evil Twins...")
    sniff(iface="wlan0", prn=detect_evil_twin, store=False)
except KeyboardInterrupt:
    print("Scan stopped.")
except PermissionError:
    print("Run the script with sudo or admin privileges.")