M1z23R/ESP8266-EvilTwin #32: how to run the ino program on esp32 wroom

mandabenjamin, 2022-01-28:
if I use esp8266 it works fine, I tried using esp32 wroom why can't it? is there a solution

mandabenjamin, 2022-02-13:
> kayanya belum bisa kalau pada esp32 ..work esp8266 dan wemos...padahal diklaim esp32 lebih powerfull ... tapi intruksi yah lum dapat

Kalo pake esp8266 saya sudah bisa kak untuk coding dengan logo isp indo,,sperti indihime dll,dan 1 alat bisa untuk deauther dan eviltwin,kekurangannya kalo esp8266 jarak tangkapan kurang jauh meski sudah d tambah antena dan kurang ekspresif,makanya saya masih mencoba dengan esp32

cakralithium, 2022-02-14:
bnr bang emang esp32 tuh type terbaru di cap lebih powerfull daripada esp8266 ... dengan wemos d1 mini pro plus xteenal aja masih klah dengan esp8266 apalagi drngan esp32.. midah mudah bisa terealisasikam demgam eap32 yah bamg