M1z23R/ESP8266-EvilTwin #33: its work

cakralithium, 2022-01-29:
intruksi yang simple tapi mudah dipahami.. dengan pengembangan intruksi jadi tambah bernilai lebih hanya ya hanya dengan 1 modul.. 

great..

Bayezid082, 2022-01-29:
Give give me bin file plz

Bayezid082, 2022-01-29:
[image]

yunstech, 2022-02-02:
> intruksi yang simple tapi mudah dipahami.. dengan pengembangan intruksi jadi tambah bernilai lebih hanya ya hanya dengan 1 modul..
> 
> great..

bagi file bin nya dong

cakralithium, 2022-02-05:
kan udah ada source code yah bang.... di youtube sudah ada coba cek chanel youtube work from home 2021 disitu sudah ada file bin dan ino

mandabenjamin, 2022-02-13:
[image]