M1z23R/ESP8266-EvilTwin #43: Customizable HTML Pages

Ni3ru, 2023-10-20:
Is there a chance that you could add a feature like you can upload customizable Html pages then delete them afterwards(To reduce storage load on esp)?

If yes, could you provide us the html page sample so we can customize it on our own?

I think this suggestion would be great if implemented.

Thanks.

OP-EX, 2023-10-28:
> Is there a chance that you could add a feature like you can upload customizable Html pages then delete them afterwards(To reduce storage load on esp)?
>
> If yes, could you provide us the html page [...]

what do you mean

Ni3ru, 2023-10-28:
> what do you mean

Like customizable phishing html page

OP-EX, 2023-10-28:
> Like customizable phishing html page

[Login Page.txt](https://github.com/M1z23R/ESP8266-EvilTwin/files/13196424/Login.Page.txt)

Ni3ru, 2023-10-28:
Thanks. but in which line of code should I put that code? I'm sorry I am still learning this

Ni3ru, 2023-10-28:
and what I really mean is uploading customizable html phishing page on web interface(like upload and delete button on a browser) But thanks to that code you sent I will spent my free time to learn it.

OP-EX, 2023-10-28:
> and what I really mean is uploading customizable html phishing page file on web interface(like upload and delete button on a browser) But thanks to that code you sent I will spent my free time to learn it.

this may help
google how to upload html to esp8266

this is a video on how to 
https://www.youtube.com/watch?v=QLGwI5tC9yk

M1z23R, 2023-10-29:
@Ni3ru I am working as both full time software engineer (8h/day) and I am also freelancing (at least 4h/day) so I can't find the time.
However, I am working on a new version as we speak.
Some of the features will be:

- Improved logic
- Default admin/phishing page
- Ability to upload new pages through sketch data upload
- Ability to upload new (1 at a time) pages through admin panel
- And potentially more

HofficeCleanex, 2023-12-13:
> @Ni3ru I am working as both full time software engineer (8h/day) and I am also freelancing (at least 4h/day) so I can't find the time. However, I am working on a new version as we speak. Some of the [...]

me avisas si pued ayudarte en algo bro ya que tambien , el codigo que proporcionaste lo estoy fusionando para que sea multiple, a que me refiero a que realice hotspot para redes sociales y redes wifi, aqui adjunto un boceto
[image]

DrPl4g4, 2024-12-06:
ya esta para html customizable